dojohnso/OctoPrint-PrintJobHistory #5: Please add a timeout for the "print finished" popup

hakan42, 2023-11-28:
Hi,

I have a few multi-day prints going on. I'd like to be able to have octopi shut down itself, but at the same time, don't want to lose the information in the "print finished" popup.

Could this be automatically saved, perhaps selectable by an option?

Regards,
Hakan

dojohnso, 2023-11-29:
Hi @hakan42! I actually have that print finished popup disabled and haven't noticed anything missing, but I haven't looked too close. Is there info in that pop up that doesn't get saved to the history list? If so, please  let me know and I can look into this.

hakan42, 2023-11-29:
Actually, I had only one print finished with your version of the plugin. In that one, I could see the thumbnail, but not the picture from the webcam.

If picture could be in there automatically, nothing else would be needed.

I'll have another print running which should be finished tonight, I'll check the behavior on finish once more with that one.

dojohnso, 2023-11-29:
Just curious, what are your settings for the Camera tab? 

[image]

hakan42, 2023-11-29:
Ah, just while taking a screenshot for the same, I noticed the "If both ..." dropdown. I just changed that to "Camera" and saved. I hope that this setting will be active for the current running print already, if not, I will run a small print tomorrow to verify.

[image]

dojohnso, 2023-11-30:
> Ah, just while taking a screenshot for the same, I noticed the "If both ..." dropdown. I just changed that to "Camera" and saved. I hope that this setting will be active for the current running print already, if not, I will run a small print tomorrow to verify.
> 
> 
> 
> But the checkbox "Use thumbnail after print, instead of camera snapshot" and the next dropdown "If both ... which one should be preferred" seem to contradict each other, maybe the text could be a bit clearer ?
> 
> 
> 
> [image]
> 
> 

Ok cool let me know how it goes. 

Yeah I agree on the wording there. I'll take a look at what those options are doing and get that cleaned up and/or make it clearer. Thanks!