dojohnso/OctoPrint-PrintJobHistory #10: Can't activate Spool Manager (version 1.7.0)

dojohnso, 2023-11-29:
per @jjnether https://github.com/OllisGit/OctoPrint-PrintJobHistory/issues/217

> When trying to select filament tracking, the option for spool manager that I have installed is greyed out (see below). I even get > > notifications that I have the option of activating something for tracking filament.
> 
> [image]
> 
> Spool Manager version 1.7.0
> Print Job History version 1.17.0
> 
> Any help would be appreciated!

dojohnso, 2023-11-29:
same for me actually. i'll look into this.

Sembazuru, 2023-11-30:
Odd. I have SpoolManager v1.7.0 installed and that radio-button is selected for me. Maybe because I had it already selected when whatever broke, broke? I do get warnings about not being able to load filament information for T0 even though I have a spool selected for Tool 0 (and it's usage is accounted for in SpoolManager). Should I open a new ticket for the error?
[image]

dojohnso, 2023-11-30:
> Odd. I have SpoolManager v1.7.0 installed and that radio-button is selected for me. Maybe because I had it already selected when whatever broke, broke? I do get warnings about not being able to load filament information for T0 even though I have a spool selected for Tool 0 (and it's usage is accounted for in SpoolManager). Should I open a new ticket for this error?
> 
> [image]
> 
> FWIW, I was having this issue before you picked up PrintJobHistory... I just never reported it because OllisGit seemed to be incommunicado.

Yeah if you don't mind, open a new one so I can track it appropriately. 

And yeah, they definitely stopped paying attention a while ago. It took me months to get octoprint to acknowledge the need to take this over so there is some catching up to do. I have only made changes for the thumbnail issue so far, but will be working through other issues to get this patched up. Thank you!

dojohnso, 2023-12-14:
this seems to be resolved now that I adopted and updated both this and SpoolManager. I'm going to close this for now but will reopen if it persists on the new repos

closed 2023-12-14 by dojohnso