dojohnso/OctoPrint-PrintJobHistory #7: Print job history discards transparent background of original PNGs

eschreiner, 2023-11-29:
First of all, @dojohnso thank you so much for working on this plugin!

When viewing the print job history with a dark theme, the embedded images (@dojohnso thank you for fixing this!!!) are converted to JPGs with white background.

[image]

Not a big deal, but would be a nice improvement.

Best regards!

dojohnso, 2023-11-30:
@eschreiner thanks again for this. I've pushed a fix in the pre-release channel if you're interested in checking it out. I'll likely push it out to production later today or tomorrow.

closed 2023-11-30 by dojohnso

eschreiner, 2023-12-01:
have just installed the DEV version - much better! Thank you so much!

[image]