dojohnso/OctoPrint-PrintJobHistory #16: Minor spelling issue in dialog?

EricZwirnmann, 2023-12-14:
[image]

Think it should say "Spool Vendor" 

P.S.
 (actually, Vendor is British English, Vender is Am. English).
Sorry for the nit-pick, I write code for a living and these things nag me too. :-)

dojohnso, 2023-12-14:
Oh I know _exactly_ what you mean. 😅

Good catch. I'm cleaning things up like this when I see them. I'll be sure to fix this one up too. Thanks!

dojohnso, 2023-12-14:
ok this is weird, and might just be a template cache thing. I saw the "Vendorss" in my local version, scoured the code and couldn't find any reason for it. I fixed a couple other minor things and rebuilt and reinstalled locally and it's fixed. So  I'm going to close this for now. Let me know if it doesn't go away after the next update though. (no sure yet when that will be)

[image]

closed 2023-12-14 by dojohnso