dojohnso/OctoPrint-PrintJobHistory #23: CSV Export not working

yenk7, 2024-05-31:
Upon clicking the CSV Export button, the webpage indicates that the link is loading but perpetually keeps buffering. The issue occurred in OctoPrint version 1.10.1. While CSV Export legacy functions, user column information is missing from the CSV in this mode.

Beastly-Kitty, 2024-06-29:
also ran into this recently. data is present but export function no longer works from any page.

jmarc67, 2024-08-19:
Same here, export function no longer works

RalfZi, 2024-11-09:
Same by me
sqlite3.ProgrammingError: SQLite objects created in a thread can only be used in that same thread.

`sqlite3.ProgrammingError: SQLite objects created in a thread can only be used in that same thread. The object was creat> Nov 09 14:34:12 octoprint kernel: workqueue: drm_fb_helper_damage_work hogged CPU for >13333us 256 times, consider switching to WQ_UNBOUND Nov 09 14:35:01 octoprint CRON[8463]: pam_unix(cron:session): session opened for user root(uid=0) by root(uid=0) Nov 09 14:35:01 octoprint CRON[8464]: (root) CMD (command -v debian-sa1 > /dev/null && debian-sa1 1 1) Nov 09 14:35:01 octoprint CRON[8463]: pam_unix(cron:session): session closed for user root Nov 09 14:35:18 octoprint octoprint[8246]: 2024-11-09 14:35:18,893 - octoprint.server.util.tornado - ERROR - Exception in WSGI application Nov 09 14:35:18 octoprint octoprint[8246]: Traceback (most recent call last): Nov 09 14:35:18 octoprint octoprint[8246]: File "/home/.../OctoPrint/lib/python3.12/site-packages/octoprint/server/util/tornado.py", line 782, in handle_request Nov 09 14:35:18 octoprint octoprint[8246]: chunk = await loop.run_in_executor(self.executor, next_chunk) Nov 09 14:35:18 octoprint octoprint[8246]: ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ Nov 09 14:35:18 octoprint octoprint[8246]: File "/usr/lib/python3.12/concurrent/futures/thread.py", line 58, in run Nov 09 14:35:18 octoprint octoprint[8246]: result = self.fn(*self.args, **self.kwargs) Nov 09 14:35:18 octoprint octoprint[8246]: ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ Nov 09 14:35:18 octoprint octoprint[8246]: File "/home/.../OctoPrint/lib/python3.12/site-packages/octoprint/server/util/tornado.py", line 777, in next_chunk Nov 09 14:35:18 octoprint octoprint[8246]: return next(app_response_iter) Nov 09 14:35:18 octoprint octoprint[8246]: ^^^^^^^^^^^^^^^^^^^^^^^ Nov 09 14:35:18 octoprint octoprint[8246]: File "/home/.../OctoPrint/lib/python3.12/site-packages/werkzeug/wsgi.py", line 500, in __next__ Nov 09 14:35:18 octoprint octoprint[8246]: return self._next() Nov 09 14:35:18 octoprint octoprint[8246]: ^^^^^^^^^^^^ Nov 09 14:35:18 octoprint octoprint[8246]: File "/home/.../OctoPrint/lib/python3.12/site-packages/werkzeug/wrappers/response.py", line 50, in _iter_encoded Nov 09 14:35:18 octoprint octoprint[8246]: for item in iterable: Nov 09 14:35:18 octoprint octoprint[8246]: File "/home/.../OctoPrint/lib/python3.12/site-packages/octoprint_PrintJobHistory/common/CSVExportImporter.py", line > Nov 09 14:35:18 octoprint octoprint[8246]: for job in allJobsDict: Nov 09 14:35:18 octoprint octoprint[8246]: File "/home/.../OctoPrint/lib/python3.12/site-packages/peewee.py", line 4683, in next Nov 09 14:35:18 octoprint octoprint[8246]: self.cursor_wrapper.iterate() Nov 09 14:35:18 octoprint octoprint[8246]: File "/home/.../OctoPrint/lib/python3.12/site-packages/peewee.py", line 4594, in iterate Nov 09 14:35:18 octoprint octoprint[8246]: row = self.cursor.fetchone() Nov 09 14:35:18 octoprint octoprint[8246]: ^^^^^^^^^^^^^^^^^^^^^^ Nov 09 14:35:18 octoprint octoprint[8246]: sqlite3.ProgrammingError: SQLite objects created in a thread can only be used in that same thread. `

vojtakaniok, 2025-02-24:
@yenk7  @rtrout @Beastly-Kitty @jmarc67 @RalfZi  should be fixed in my new release, feel free to update

rtrout, 2025-02-25:
Thanks, works for me.